stm32duino/VL53L4CD #5: InitSensor(); return error

AlexLazarovich, 2022-05-11:
In my setup when I call InitSensor(); for my sensor it returns an error all the time, even if I put it in a loop, I've tried going into the header files a bit, inside the InitSensor method, it returns an error from the VL53L4CD_SensorInit() method, further than that I didn't get into.
It won't even run when I upload one of your examples.

fpistm, 2022-05-11:
Hi
Could you share your setup ?
Core version, board, ...

AlexLazarovich, 2022-05-12:
Hey,
I somehow got past that error with one sensor, but my board is going to use 2 sensors on the same i2c.
I'd love it if you could maybe even upload an example with 2 or more sensors, how you would init them and use them afterwards.
When I try to use 2 of them, weird errors occur, but I'm almost sure it's because of me, since I'm probably not using them correctly. 
I'm uploading my init of the sensors, it won't even go to loop as far as I can see, maybe I'm doing something out of order.

I'm using an Arduino nano 33 ble.

[image]
[image]

cparata, 2022-05-12:
Hi @AlexLazarovich ,
[here](https://github.com/stm32duino/X-NUCLEO-53L4A1/blob/main/examples/X_NUCLEO_53L4A1_HelloWorld/X_NUCLEO_53L4A1_HelloWorld.ino) you can find how to do the setup of more than one sensor for VL53L4CD. Are pin "A1" and pin "A2" respectively the XSHUT pins for dSensorL1 and dSensorL2? It's important that the XSHUT pins are well connected to the board in order to use more than one ToF sensor. Finally, please try to change 0x53 I2C address (with for example 0x10) for dSensorL1. The I2C addresses 0x52 and 0x53 are already used by dSensorL2 respectively for write operations and read operations on I2C bus.
Best Regards,
Carlo

AlexLazarovich, 2022-05-12:
Hey, the XSHUT pins are connected correctly, the address change did the job. I didn't know that it uses 2 addresses, that is probably why my Serial kept crashing after I've init-ed them with 0x52 and 0x53, there was a collision on 0x53, I'd love it if you could add that remark, that when adding more than one I2C user in general, keep the next address available.

Also, for the original problem of the issue, if someone would find it helpful in the future, my problem in advance was, that when I designed the board I've routed 5v to the sensors, so I've had to add a step-down level shifter (chose it instead of flying wires at first) which dropped down the voltage to about 3.05v which should be a ok for the sensor, but my XSHUT was routed correctly to an analog output pin on my arduino nano 33 ble, which was outputting 3.3v, and the sensor would not init.
When I connected the sensor with 3.3v from my nano 3.3v voltage output it miraculasly came to life.
My only guess is that the difference in voltages between the XSHUT input and the VDD input just didn't let it work.
May my experience help someone in the future.

Thank you for your help!
Alex.

closed 2022-05-12 by AlexLazarovich

cparata, 2022-05-12:
Hello @AlexLazarovich ,
actually the info about the I2C address is contained in the datasheet of the component [here](https://www.st.com/resource/en/datasheet/vl53l4cd.pdf) paragraph 3: 

"A message contains a series of bytes preceded by a start condition, and followed by either a stop or repeated
start (another start condition but without a preceding stop condition), followed by another message. The first byte
contains the device address (0x52) and also specifies the data direction. If the least significant bit is low (that
is, 0x52) the message is a master-write-to-the-slave. If the LSB is set (that is, 0x53) then the message is a
master-read-from-the-slave."

Thanks to have shared your tips.
I will close this issue.
Best Regards,
Carlo