stm32duino/VL53L4CD #6: Setting Up Multiple Sensors

Hs-Cheng, 2022-05-25:
Hi,

My problem is similar to another issue that was closed two weeks ago. I am trying to setup two VL53L4CD sensors together, and I would list my questions below.

1. Why it is important to connect the XSHUT pins when setting up multiple sensors? Can I just let them floating?
2. During the setup phase, both the XSHUT pins should be "HIGH" to keep the sensors turn on, is that correct?
3. By reading another issue mentioned above, I've already known that each sensor occupies two I2C addresses (for example, 0x52 and 0x53). Are there other restrictions when setting up the addresses? Can I just use some random numbers that are not continuous like [0x12, 0x28]?

I also attached the code of:
1. Declaring two instances of the sensor object. 
2. The setup function of my Arduino code.

Thank you very much for your assistance.

All the best,
HC

[image]

cparata, 2022-05-25:
Hello @Hs-Cheng ,
the X-SHUT pin is fundamental if you use multiple sensors because it avoids conflicts on the I2C bus. When you power the sensors, all of these sensors have the same I2C address (0x52), so if they are connected to the same I2C bus you get an hardware conflict. To avoid it, you must disable all the sensors at the beginning (putting all the XSHUT pins to LOW). Then, you can initialize the sensors one by one as you correctly did in your sketch. If you are using just one sensor, you cannot leave the XSHUT pin floating but at least connected to a Pull-Up resistor to keep the sensor active.
Anyway, your sketch seems fine to me. Doesn't it work? Did you connect correctly all the pins?
Best Regards,
Carlo

cparata, 2022-05-25:
Sorry  @Hs-Cheng ,
now I found an issue in your sketch. You are using the same XSHUT pins on the 2 sensors (pin 17). This is wrong. You must assign a different XSHUT pin to the 2 sensors otherwise you are not able to control them in an independent way.
Try for example to assign pin 17 as XSHUT pin for sensor 1 and pin 18 as XSHUT pin for sensor 2.
Best Regards,
Carlo

Hs-Cheng, 2022-05-25:
Hi @cparata ,

Thank you for your reply. I have several follow-up questions (sorry I am not experienced in I2C).

1. In my setup function above, when should I turn off sensor 2? Before the InitSensor() function of sensor 1 or before the begin() function of sensor 1? Also, if I need to turn off sensor 2 when initializing sensor 1, is it better to separate the setup codes for sensor 1 and sensor 2? Right now the setup codes for both sensors are mixed together.
2. Is there a way to modify the I2C address by hardware wiring, instead of software coding? This is the datasheet of an accelerometer I used before: https://cdn-learn.adafruit.com/downloads/pdf/adxl343-breakout-learning-guide.pdf                    
On page 9 and 10, it says that we can connect 3.3V to the SDO pin to modify the I2C address. Is there a similar function on the VL53L4CD sensor?

Thank you again for your assistance.

All the best,
HC

cparata, 2022-05-25:
Hello @Hs-Cheng ,
you need to turn off all the sensors before starting to call the InitSensor() in cascade. So, as you did in the sketch, you must call begin and VL53L4CD_Off for all sensors before and then you can call the InitSensor() one by one.
Unfortunately, for VL53L4CD component you can set the I2C address by software. There is not any mechanism to change it by hardware. Anyway, now is your setup working? Can I close this issue?
Best Regards,
Carlo

Hs-Cheng, 2022-05-25:
Hi @cparata ,

My setup is working now. Thank you very much for your help. I am going to close this issue.

All the best,
HC

closed 2022-05-25 by Hs-Cheng

nh1628, 2023-06-02:
@Hs-Cheng  I'm currently struggling with getting 2 of the sensors to read together, would it be possible to post your code so I can see how you got it working?

Hs-Cheng, 2023-06-02:
Hi @nh1628,

I am glad that you solved the problem by modifying the I2C address. Usually the address of each sensor need to be separated by at least 2. Therefore, I guess 0x30 and 0x28 will also work if 0x30 is not occupied by other sensors/ICs.

I wish you all the best with your project.
HC