stm32duino/VL53L4CD #8: Not working with ESP32-S3

Tylersuard, 2023-05-30:
Hello.  I am having some issues with using this code with my ESP32-S3 breakout board from Adafruit (QTPY stemma)

When I upload the example script, the board just acts as a USB drive and won't actually do anything.

Here is the code I am using:

#include <Arduino.h>
#include <Wire.h>
#include <vl53l4cd_class.h>
#include <string.h>
#include <stdlib.h>
#include <stdio.h>
#include <stdint.h>
#include <assert.h>
#include <stdlib.h>

#define DEV_I2C Wire
#define SerialPort Serial

// Components.
VL53L4CD sensor_vl53l4cd_sat(&Wire, 0);

/* Setup ---------------------------------------------------------------------*/

void setup()
{

  // Initialize serial for output.
  SerialPort.begin(115200);
  SerialPort.println("Starting...");

  // Initialize I2C bus.
  DEV_I2C.begin();

  // Configure VL53L4CD satellite component.
  sensor_vl53l4cd_sat.begin();

  // Switch off VL53L4CD satellite component.
  sensor_vl53l4cd_sat.VL53L4CD_Off();

  //Initialize VL53L4CD satellite component.
  sensor_vl53l4cd_sat.InitSensor(0x29);

  // Program the highest possible TimingBudget, without enabling the
  // low power mode. This should give the best accuracy
  sensor_vl53l4cd_sat.VL53L4CD_SetRangeTiming(200, 0);

  // Start Measurements
  sensor_vl53l4cd_sat.VL53L4CD_StartRanging();
}

void loop()
{
  uint8_t NewDataReady = 0;
  VL53L4CD_Result_t results;
  uint8_t status;
  char report[64];

  do {
    status = sensor_vl53l4cd_sat.VL53L4CD_CheckForDataReady(&NewDataReady);
  } while (!NewDataReady);

  if ((!status) && (NewDataReady != 0)) {
    // (Mandatory) Clear HW interrupt to restart measurements
    sensor_vl53l4cd_sat.VL53L4CD_ClearInterrupt();

    // Read measured distance. RangeStatus = 0 means valid data
    sensor_vl53l4cd_sat.VL53L4CD_GetResult(&results);
    snprintf(report, sizeof(report), "Status = %3u, Distance = %5u mm, Signal = %6u kcps/spad\r\n",
             results.range_status,
             results.distance_mm,
             results.signal_per_spad_kcps);
    SerialPort.print(report);
  }

}

Tylersuard, 2023-05-30:
@fpistm Thank you for your quick response.  How would I do that?  I can't even get this code to run.

Tylersuard, 2023-05-30:
@fpistm I am getting the response: 255

Tylersuard, 2023-05-30:
I am using this code.  After compiling and uploading, I reset the board using the button.  Now the ESP32-S3 QTPY board will stay on for a second, then turn off, then turn on again, then turn off again, etc.

#include <Arduino.h>
#include <Wire.h>
#include <vl53l4cd_class.h>
#include <string.h>
#include <stdlib.h>
#include <stdio.h>
#include <stdint.h>
#include <assert.h>
#include <stdlib.h>

#define DEV_I2C Wire
#define SerialPort Serial

// Components.
VL53L4CD sensor_vl53l4cd_sat(&Wire, 0);

/* Setup ---------------------------------------------------------------------*/

void setup()
{

  // Initialize serial for output.
  SerialPort.begin(115200);
  SerialPort.println("Starting...");

  // Initialize I2C bus.
  DEV_I2C.begin();

  // Configure VL53L4CD satellite component.
  sensor_vl53l4cd_sat.begin();

  //Initialize VL53L4CD satellite component.
  sensor_vl53l4cd_sat.InitSensor(0x29);

}

void loop()
{
  VL53L4CD_Result_t results;
  uint8_t status;
  char report[64];

    // Read measured distance. RangeStatus = 0 means valid data
  SerialPort.print("results:");
  SerialPort.print(sensor_vl53l4cd_sat.VL53L4CD_GetResult(&results));
  snprintf(report, sizeof(report), "Status = %3u, Distance = %5u mm, Signal = %6u kcps/spad\r\n",
             results.range_status,
             results.distance_mm,
             results.signal_per_spad_kcps);
    SerialPort.print(report);
  }

This is the printout from the serial monitor:

1Status =  31, Distance = 16333 mm, Signal =   1033 kcps/spad
results:1Status =  31, Distance = 45887 mm, Signal =    220 kcps/spad
results:1Status =  31, Distance = 45887 mm, Signal =    220 kcps/spad
results:1Status =  31, Distance = 16333 mm, Signal =   1033 kcps/spad
results:1Status =  31, Distance = 16333 mm, Signal =   1033 kcps/spad
results:

From what I understand, status of 1 means invalid reading?

fpistm, 2023-05-31:
> @fpistm I am getting the response: 255

It is timeout error code. Looking at the code, I guess the @ you used is not correct.
You gave 0x29 but I think it is not correct. By default @ is 0x52 which is on 8 bits but later encoded on 7 bits: 0x52>>1 = 0x29.
So, you should probably do:
```patch
- sensor_vl53l4cd_sat.InitSensor(0x29);
+ sensor_vl53l4cd_sat.InitSensor();
```

cparata, 2023-05-31:
> > @fpistm I am getting the response: 255
> 
> It is timeout error code. Looking at the code, I guess the @ you used is not correct. You gave 0x29 but I think it is not correct. By default @ is 0x52 which is on 8 bits but later encoded on 7 bits: 0x52>>1 = 0x29. So, you should probably do:
> 
> ```diff
> - sensor_vl53l4cd_sat.InitSensor(0x29);
> + sensor_vl53l4cd_sat.InitSensor();
> ```

Hello @fpistm ,
if the hardware setup is correct, it should work also with
`sensor_vl53l4cd_sat.InitSensor(0x29);`
because it is like to assign to the sensor the 0x29 address. I suspect some issues in the hardware setup. @Tylersuard , what is the board with VL53L4CD component that you are using?
Best Regards,
Carlo

fpistm, 2023-05-31:
@cparata I guess it is this one: https://www.adafruit.com/product/5426

fpistm, 2023-05-31:
Pullup issue? Don't know if the core used set PU by default or assumes hardware one?

cparata, 2023-05-31:
> @cparata I guess it is this one: https://www.adafruit.com/product/5426

This is the motherboard. I'm referring to the daughterboard with the VL53L4CD. I would like to understand better how the component is connected to the MCU.

Tylersuard, 2023-05-31:
The daughterboard is this: https://www.adafruit.com/product/5396

in the instructions for the board there is no wiring xshut to anything, we just plug in the Stemma connector and it should work from there.

Also I scanned for i2c addresses and 0x29 is the only one that came up.

cparata, 2023-06-01:
Hello @Tylersuard ,
now the hardware setup is clear. The XSHUT pin is connected directly to Vdd in the daughterboard and also I2C Pull-Up resistors are mounted in the daughterboard. So, I only suggest these modifications in the source code that you posted.
I would replace:
`VL53L4CD sensor_vl53l4cd_sat(&Wire, 0);`
with
`VL53L4CD sensor_vl53l4cd_sat(&Wire, -1);`
and, as suggested by @fpistm , I would replace:
`sensor_vl53l4cd_sat.InitSensor(0x29);`
with
`sensor_vl53l4cd_sat.InitSensor();`
Let me know if in this way something changes.
Best Regards,
Carlo

cparata, 2023-06-05:
Hello @Tylersuard ,
any update?
Carlo

fpistm, 2023-07-11:
Close it as no feedback from OP about the suggested modifications of the sketch.

closed 2023-07-11 by fpistm