stm32duino/VL53L4CD #13: VL53L4CD_GetResult Divide by Zero

MarksWizard, 2024-10-30:
**Describe the bug**
There is a potential divide by zero issue in the VL53L4CD_GetResult function of vl53l4cd_api.cpp.

**To Reproduce**
This is the original function.  If number_of_spad = 0, then signal_rate_kcps and ambient_rate_kcps will be divided by zero.
```
VL53L4CD_ERROR VL53L4CD::VL53L4CD_GetResult(
  VL53L4CD_Result_t *p_result)
{
  VL53L4CD_ERROR status = VL53L4CD_ERROR_NONE;
  uint16_t temp_16;
  uint8_t temp_8;
  uint8_t status_rtn[24] = { 255, 255, 255, 5, 2, 4, 1, 7, 3,
                             0, 255, 255, 9, 13, 255, 255, 255, 255, 10, 6,
                             255, 255, 11, 12
                           };

  status |= VL53L4CD_RdByte(dev, VL53L4CD_RESULT_RANGE_STATUS,
                            &temp_8);
  temp_8 = temp_8 & (uint8_t)0x1F;
  if (temp_8 < (uint8_t)24) {
    temp_8 = status_rtn[temp_8];
  }
  p_result->range_status = temp_8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SPAD_NB,
                            &temp_16);
  p_result->number_of_spad = temp_16 / (uint16_t) 256;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SIGNAL_RATE,
                            &temp_16);
  p_result->signal_rate_kcps = temp_16 * (uint16_t) 8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_AMBIENT_RATE,
                            &temp_16);
  p_result->ambient_rate_kcps = temp_16 * (uint16_t) 8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SIGMA,
                            &temp_16);
  p_result->sigma_mm = temp_16 / (uint16_t) 4;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_DISTANCE,
                            &temp_16);
  p_result->distance_mm = temp_16;

  if (number_of_spad == 0){
	p_result->signal_per_spad_kcps = 0;
	p_result->ambient_per_spad_kcps = 0;
	return status;}

  p_result->signal_per_spad_kcps = p_result->signal_rate_kcps
                                   / p_result->number_of_spad;
  p_result->ambient_per_spad_kcps = p_result->ambient_rate_kcps
                                    / p_result->number_of_spad;

  return status;
}
```

**Expected behavior**
My code was occasionally throwing a divide by zero fault, which causes a panic and resets the device on ESP32.  After seeing it happen enough times, I decided to investigate further.

**Desktop (please complete the following information):**
 - OS: Mac OSX Sonoma 14.6.1
 - Arduino IDE version: 2.3.3
 - Library version: 1.0.4

**Hardware (please complete the following information):**
 - Board Name: ESP32 DevKitC-1 
 - Hardware Revision: 1.1
 - Extra hardware used if any: Adafruit VL53L4CD 
 
**Additional context**
The issue is easily resolved by checking to see if number_of_spad is 0 before doing the division.

```
VL53L4CD_ERROR VL53L4CD::VL53L4CD_GetResult(
  VL53L4CD_Result_t *p_result)
{
  VL53L4CD_ERROR status = VL53L4CD_ERROR_NONE;
  uint16_t temp_16;
  uint8_t temp_8;
  uint8_t status_rtn[24] = { 255, 255, 255, 5, 2, 4, 1, 7, 3,
                             0, 255, 255, 9, 13, 255, 255, 255, 255, 10, 6,
                             255, 255, 11, 12
                           };

  status |= VL53L4CD_RdByte(dev, VL53L4CD_RESULT_RANGE_STATUS,
                            &temp_8);
  temp_8 = temp_8 & (uint8_t)0x1F;
  if (temp_8 < (uint8_t)24) {
    temp_8 = status_rtn[temp_8];
  }
  p_result->range_status = temp_8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SPAD_NB,
                            &temp_16);
  p_result->number_of_spad = temp_16 / (uint16_t) 256;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SIGNAL_RATE,
                            &temp_16);
  p_result->signal_rate_kcps = temp_16 * (uint16_t) 8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_AMBIENT_RATE,
                            &temp_16);
  p_result->ambient_rate_kcps = temp_16 * (uint16_t) 8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SIGMA,
                            &temp_16);
  p_result->sigma_mm = temp_16 / (uint16_t) 4;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_DISTANCE,
                            &temp_16);
  p_result->distance_mm = temp_16;

  if (number_of_spad == 0){
	p_result->signal_per_spad_kcps = 0;
	p_result->ambient_per_spad_kcps = 0;
	return status;}

  p_result->signal_per_spad_kcps = p_result->signal_rate_kcps
                                   / p_result->number_of_spad;
  p_result->ambient_per_spad_kcps = p_result->ambient_rate_kcps
                                    / p_result->number_of_spad;

  return status;
}
```

fpistm, 2024-10-30:
Hi @MarksWizard 
Thanks for the issue and detailed analysis.
Could you submit a PR, please?

MarksWizard, 2024-10-30:
Wow thanks for the quick response from both of you!  I added some lines of code to print out the results as they're read in the function.   

This is what a normal read looks like:
`09:52:37.280 -> Range Status: 0 Number of Spad: 203 Signal rate: 7120 Ambient rate: 1064 Sigma mm: 1 Distance mm: 248
`

 Here are the results from a few times it threw the error.
```
10:14:37.975 -> Range Status: 0 Number of Spad: 0 Signal rate: 2480 Ambient rate: 2632 Sigma mm: 4 Distance mm: 129

10:15:57.123 -> Range Status: 0 Number of Spad: 0 Signal rate: 2376 Ambient rate: 3080 Sigma mm: 3 Distance mm: 127

10:17:34.713 -> Range Status: 255 Number of Spad: 0 Signal rate: 6936 Ambient rate: 6936 Sigma mm: 216 Distance mm: 867
```

I am running my own complex code (2800 lines + various libraries), but using primarily the example code to initialize and read the TOF sensor.  I am not sure if I ever saw the error when running only the example code though.  I have a SHT30 Temp/Humidity sensor sharing the same lines.  I admittedly have struggled over the past few days trying to get this sensor to initialize/stay working correctly, while the SHT30 never had an issue.  Fortunately, this morning I realized that I was powering the sensors with 5v, but using the 3.3V GPIO of the ESP32 to communicate.  Switching the power source to 3.3V for the sensors seems to have stopped all of the issues.  

So it is likely that it was a hardware issue causing the number_of_spad = 0 problem, however I still think it's good to check for the zero before any division.  I would much rather have the sensor give me a bad reading, that I can check for with code,  instead of resetting the entire system.

MarksWizard, 2024-10-30:
How about this for the proposed change?

`VL53L4CD_ERROR VL53L4CD::VL53L4CD_GetResult(
  VL53L4CD_Result_t *p_result)
{
  VL53L4CD_ERROR status = VL53L4CD_ERROR_NONE;
  uint16_t temp_16;
  uint8_t temp_8;
  uint8_t status_rtn[24] = { 255, 255, 255, 5, 2, 4, 1, 7, 3,
                             0, 255, 255, 9, 13, 255, 255, 255, 255, 10, 6,
                             255, 255, 11, 12
                           };

  status |= VL53L4CD_RdByte(dev, VL53L4CD_RESULT_RANGE_STATUS,
                            &temp_8);
  temp_8 = temp_8 & (uint8_t)0x1F;
  if (temp_8 < (uint8_t)24) {
    temp_8 = status_rtn[temp_8];
  }
  p_result->range_status = temp_8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SPAD_NB,
                            &temp_16);
  p_result->number_of_spad = temp_16 / (uint16_t) 256;

  if (p_result->number_of_spad == 0){
 	p_result->range_status = 255;
 	p_result->signal_rate_kcps = 0;
  	p_result->ambient_rate_kcps = 0;
	p_result->sigma_mm = 0;
 	p_result->distance_mm = 0;
	p_result->signal_per_spad_kcps = 0;
	p_result->ambient_per_spad_kcps = 0;
	return status;
}

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SIGNAL_RATE,
                            &temp_16);
  p_result->signal_rate_kcps = temp_16 * (uint16_t) 8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_AMBIENT_RATE,
                            &temp_16);
  p_result->ambient_rate_kcps = temp_16 * (uint16_t) 8;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_SIGMA,
                            &temp_16);
  p_result->sigma_mm = temp_16 / (uint16_t) 4;

  status |= VL53L4CD_RdWord(dev, VL53L4CD_RESULT_DISTANCE,
                            &temp_16);
  p_result->distance_mm = temp_16;

  p_result->signal_per_spad_kcps = p_result->signal_rate_kcps
                                   / p_result->number_of_spad;
  p_result->ambient_per_spad_kcps = p_result->ambient_rate_kcps
                                    / p_result->number_of_spad;

  return status;
}`

or the if statement can probably just be simplified to this?

```
if (p_result->number_of_spad == 0){
 	p_result->range_status = 255;
	return status;
}
```

cparata, 2024-10-31:
Hello @MarksWizard ,
now it's clear. The VL53L4CD cannot be powered at 5V because it's out of specs as you can see in the table below:

[image]

Our ToF experts told me that, if the sensor is used correctly, the number of spads should never be 0. Anyway we can add a control in the driver to avoid the division by 0.
Best Regards,
Carlo

MarksWizard, 2024-10-31:
Thank you for looking into this for me!  I understand that number of spads should never be 0 under normal operation, but strange things do occur from time to time and a sensor error is much better than a system crash.

I am using the [Adafruit VL53L4CD module](https://www.adafruit.com/product/5396) which has all of the components necessary to make the sensor 3.3V/5V friendly.  However the issue of using 5V power and 3.3V logic was easily overlooked because the SHT30 sensor continued to work flawlessly.

“The sensor is small and easy to use in any robotics or interactive project. Since it needs 2.8V power and logic we put the little fellow on a breakout board with a regulator and level shifting. You can use it with any 3-5V power or logic microcontroller with no worries. Works great with the 3.3V logic level of a Feather or Raspberry Pi, or the 5v level of a Metro 328 or Arduino Uno, this breakout is ready to work with most common microcontrollers or SBCs. and since it speaks I2C, you can easily connect it up with two data wires plus power and ground.”

cparata, 2024-10-31:
Hello @MarksWizard ,
okay I got it. I released a new version of the library (v1.0.5) adding the control that you suggested.
I think that we can close this issue.
Best Regards,
Carlo

closed 2024-10-31 by cparata